simondlevy/Hackflight #1: commits

robert-b, 2016-05-24:
>No commit message

it would be very nice to get an idea what you are doing.
some sort of commit text is very helpfull for all.
the code base is nice.

best
robert

simondlevy, 2016-05-24:
Thanks, Robert -- will do!  Right now I'm working on embedding the firmware in a simulator.  Whenever I make a nontrivial change to the firmware, I test it on at least one actual vehicle, to make sure I haven't messed anything up.

robert-b, 2016-05-24:
nice.
will add a c++ sample later on.
as time is scarce at the moment.
and a simulator environment makes every shine even more.
all the best
robert

brm007, 2017-11-22:
i also added some separation to the codebase:
`static Hackflight       fc;
static ThreeDFly        model;
static Imu                  imu;
static SBUS_Receiver rc;
static RevoF4            revo(imu);
`
i'd like to use the systick interrupt to sample memsdata.
thus i pulled out the mems related calls and put them into the imu class.
will add the sampling code later on.
and yes i use some china f4 based boards.
when it flies i'll add a hackflight osd :-)

and yes - i put the cpp code into implementation files.
else the compilation is some sort of a pain.

brm007, 2017-11-22:
ok, i used the 'code' tag to insert code. not the best method.

simondlevy, 2017-11-22:
Ahh, the RevoF4 -- my first flight controller, never appreciated how ahead of its time it was till I started buying those cheap F1 boards!  I'd love to see your code.   To keep the design simple and abstract it might make sense to move your IMU interrupt code into a new Board::extras method.  That's what I did with the polling on the LadybugFC.

robert-b, 2017-11-22:
i can prep my version to move to github - no problem.
i hope to finish that on the weekend.
cheers
robert

closed 2019-07-05 by simondlevy