simondlevy/Hackflight #2: Hackflight firmware on Windows

sytelus, 2017-03-01:
I'm looking for firmware code that I can put in [AirSim](https://github.com/microsoft/airsim). Basically, this firmware would a simple cross platform library that takes in sensor inputs from the simulator and outputs rotor controls.

Is this possible to do with hackflight firmware?

brm007, 2017-03-01:
simon, this would be interesting to know.
at least my pc is outdated when i look at the unreal requirements.

simondlevy, 2017-03-01:
@Sytelus: Cool project idea!  Short answer: YES.

I've been using V-REP as the flight-simulator for Hackflight (I call the combination "HackflightSim"), but have been looking for a more faster, more realistic simulator.  AirSim may be just the ticket.  
Hackflight has a special class, Board, that provides a clean, minimal interface between the Hackflight firmware and the flight controller or simulator (inputs = IMU and R/C transmitter values; outputs=four motor values).  So, depending on the extent to which the AirSim code is interleaved with Pixhawk/APM, this could be an easy or difficult project.

I'm gonna play a bit with AirSim now to see how feasible it is to work with.  If I can get a prototype working, there should be a new commit to this repository in a few days.

simondlevy, 2017-03-01:
Okay, our Geology Department appears to have some state-of-the-art Windows 10 boxes that meet the AirSim specs (32GB RAM, GPU, SSD, etc.)  I'm gonna see if they'll let me have admin on 'em so I can install AirSim.

sytelus, 2017-03-01:
Great! I've been looking for Hackflight like architecture for few days now and it's absolutely great you have designed it in a way I have wished all firmwares were designed. I've been refactoring everything in AirSim to make sure MavLink/pixHawk parts are well isolated behind interfaces. Yesterday I found one place it tiny bit wasn't and I'm fixing it right now and push it in few hours. 

It's actually fairly straight forward to interface new firmware in AirSim. The minimum interface that firmware should provide is,

1. **Input:** array of floats representing RC controls.
2. **Input:** vector3 of angular velocity from gyro
3. **Input:** vector3 of linear acceleration from accelerometer
4. **Output**: array of rotor control signals

Firmware can also provide additional interface to consume GPS signal and express user intents such as go to location, move by velocity, return to home etc. However those are optional.

Is there any "HelloHackflight" sample code that basically does few simple thing? I thinking it would be simple console app that initializes firmware, makes a call to send it RC input and another call to get rotor control signals. If you have this "Hello" app in some form, I will definitely accelerate my effort to understand its interface.

simondlevy, 2017-03-02:
Excellent idea, Sytelus!  I will get to work tonight on "HelloHackflight".  

More good news: the head of the Geology Department (a good friend here) just gave me the thumbs up to use their machines to develop AirSim :^)

As for GPS, proximity sensing, and the like: rather than sticking them in the middle of the code as most firmwares do, I've set up an "extras" feature, whereby you implement certain [Board](https://raw.githubusercontent.com/simondlevy/hackflight/master/firmware/board.hpp) class for your particular 'copter configuration.  By default these extras methods are just stubs.  This keeps the crucial code (stabilization, mixing) simple and clean.

Stay tuned for updates ....

auturgy, 2017-08-11:
G'day Simon - I'm assuming you gave Microsoft permission to use and relicense your code without attribution, as it's now bundled into AirSim as SimpleFlight! (it would be somewhat unusual for GPLv3 code to be sitting within an MIT licensed stack).

closed 2019-07-05 by simondlevy