simondlevy/Hackflight #3: .dll file creation

baymaximus, 2017-05-17:
Hi! I tried creating the .dll file in microsoft visual studio professional 2015, by following the steps as mentioned by you. But each time, I encounter the error: C1083	Cannot open include file: 'v_repExt.h': No such file or directory	v_repExtHackflight	F:\Haptics\hackflight-master\sim\v_repExtHackflight.cpp	
What do I do to overcome it??

simondlevy, 2017-05-18:
Yeah, that's a problem with your include-path in VisualStudio2015.  First, make sure you've got the latest hackflight repo (I just updated it 01:21 EST Thurs 18 May 2017).  Then check out this [video](https://youtu.be/EXX7fY1oPF4) to ensure your include-path is set appropriately.

closed 2017-05-18 by simondlevy

baymaximus, 2017-05-18:
Still not working for me.. Can it be because of my OS Windows 10??