simondlevy/Hackflight #17: Butterfly board support

juangallostra, 2018-04-17:
Develop the required code so that Hackflight can run on a [Butterfly board](https://www.tindie.com/products/TleraCorp/butterfly-stm32l433-development-board/?pt=ac_prod_search).

juangallostra, 2018-05-08:
@simondlevy : Although we haven't tried yet to fly a real drone with the butterfly board I think that we can already say that Hackflight does support the butterfly board. Hence, I believe we can close this issue.

File that adds support for the butterfly board: https://github.com/simondlevy/Hackflight/blob/master/src/boards/butterfly.hpp

simondlevy, 2018-05-09:
Reopened pending investigation of PWM output on digital pins 5,6.

kriswiner, 2018-05-09:
Whats' the issue with 5 and 6?

simondlevy, 2018-05-10:
By [modifying](https://github.com/simondlevy/grumpyoldpizza/blob/master/stm32l4/variants/STM32L433CC-Butterfly/variant.cpp#L48-L55) my version of STM32L4  I can now spin motors on pins 3,4,5,6.

Before closing this issue I will test SBUS receiver.

juangallostra, 2018-09-04:
@simondlevy I guess we can close this issue now? :smiley:

closed 2020-08-26 by simondlevy