simondlevy/Hackflight #38: sentral quaternion update rate

moritzmaier, 2019-02-01:
I was wondering what the update rate of the kalman filter inside the em7180 really is!? In the manual of the em7180 it says something about 30Hz, which seems a little bit slow for quadrotor attitude control. I understand that the Q_Rate_Divisor sets the speed of the quaternion readings based on the gyro update rate, e.g. 1/5 of 330Hz as in sentral.hpp. But that can also mean that the quaternion I get at 330/5 = 66Hz stays the same for ~2 update cycles. Am I missing something here?

doesn'tything about the real update rate

kriswiner, 2019-02-02:
The Kalman filter is one f several filters usedin the SENtral algorithm.
The quaternions are updated at the rate of the gyro (or less with qrate
divisor), the Kalman filter is at a fixed 30 Hz. I would recommend running
the gyro at 800 Hz or 1 kHz and the quats at 400 Hz update rate for a UAV
application.

moritzmaier, 2019-02-02:
Thanks for the fast reply. That means

GYRO_RATE = 800 and
Q_RATE_DIVISOR = 1

or 

Q_RATE_DIVISOR = 0 ?

I thought the maximum gyro rate that can be handled by the EM7180 was 400Hz!?

closed 2020-08-26 by simondlevy