simondlevy/Hackflight #39: error compilation

yann-insa, 2019-03-29:
hi Simond,
I have a little problem on arduino mac v1.8.5 and 1.8.9, I can't compile your superfly, I can't understand why, I add librairy as your wiki.
Can you help me, you will find in attchment my debug compilator on arduino.
Thanks in advance
Regards
[error superfly.txt](https://github.com/simondlevy/Hackflight/files/3023612/error.superfly.txt)

simondlevy, 2019-03-30:
Thank you for discovering this problem, yann-insa.  I have added a fix to the repository, so please pull it and let me know whether you still have the problem.

yann-insa, 2019-04-01:
hi,
Compilation is okay now, when I'm connect on superfly wifi, serial debug indicate my MAC adress.
Howerver, in hackfright ground control station, I can't click on connect or anything.
Do I must need a game controller?I just want to read IMU data. 

`MacBook-Air-de-yann:superfly` yann$ python3 superfly.py 
Traceback (most recent call last):
  File "superfly.py", line 30, in <module>
    from pysticks import get_controller
ModuleNotFoundError: No module named 'pysticks'
`
I have no error to appear on python 3.7.3
[image]

yann-insa, 2019-04-01:
Just another question, hackfright does works with wifi connexion or UART connexion (tty.wchusbserial1410 in superfly)?
Have a good day and thanks in advance.

Best regards

simondlevy, 2019-04-01:
Everything you need to know is in the [wiki](https://github.com/simondlevy/Hackflight/wiki/S.01-SuperFly-Hardware-Setup).

yann-insa, 2019-04-01:
I use it, but I can't click on connect button on hackfright.py even if I install all libraries (pygame, pyserial, MSPPG, pystick) with serial and wifi connexion and I don't know where is my mistake.
I also test it on windows but do samething (not install python and MSPPG)

simondlevy, 2019-04-01:
I'm sorry, I can't help you with that from here.

closed 2020-08-26 by simondlevy