simondlevy/Hackflight #54: Using an Adafruit Precision NXP 9-DoF IMU with TinyPICO

dannyadu, 2020-02-26:
Would it be possible to use this [IMU](https://www.adafruit.com/product/3463) with Hackflight? Seeing as Adafruit has already made a [library](https://github.com/adafruit/Adafruit_AHRS) for the IMU it feels like would be possible, but I'm not quite sure how to go about implementing it.

simondlevy, 2020-02-26:
Yes, I see no problem using that IMU for Hackflight.  Since I like to test out all my code before posting or suggesting anything, I've ordered a couple of these IMUs from Adafruit and should have them working with Hackflight by the end of next week (6 March).  If you feel like getting started sooner than that, you can copy/paste/modify my [code](https://github.com/simondlevy/Hackflight/blob/master/src/imus/softquats/mpu9250.hpp) for working with the MPU9250 sensor.

dannyadu, 2020-02-27:
Great, thanks! I had already started off by basing my code on [USFS](https://github.com/simondlevy/Hackflight/blob/master/src/imus/usfs.hpp) using Adafruit's [AHRS](https://github.com/adafruit/Adafruit_AHRS) library for filtering and treating it as hardware quaternion data. Can I use Hackflight's filters instead and just read the raw IMU data?

simondlevy, 2020-02-27:
I was just thinking about this yesterday.  It might be good to compare Hackflight's quaternion filter to the one in Adafruit's library, to see if one gives better/smoother outputs than the other.

danieladugyan, 2020-03-01:
I've managed to get the IMU working to some extent in Hackflight (I'm testing with the GCS), both with Adafruit's [filter library](https://github.com/adafruit/Adafruit_AHRS/tree/master/src) (Madgwick, Mahony and NXP) and Hackflight's [filters](https://github.com/simondlevy/Hackflight/blob/master/src/filters.hpp) (Madgwick 6DoF and 9DoF). However, I've been getting completely unusable performance out of Adafruit's filter library (with my ESP32). The IMU runs moderately well using Hackflight's filters in the beginning, but after a few seconds it drifts away from the actual IMU position (probably because I'm not compensating for the measurement errors). 
I've also implemented MadgwickQuaternionFilter9DOF in a new [softquat.hpp](https://github.com/simondlevy/Hackflight/blob/master/src/imus/softquat.hpp) file, but it doesn't seem to have any affect on how the IMU behaves in the GCS.

danieladugyan, 2020-03-01:
I just tried with Hackflight's Mahony filter and it's giving me the best results. It looks a bit jittery in the GCS, but at least I'm not seeing any sensor drift.

simondlevy, 2020-03-01:
Thank you for the updates, Daniel.  My IMUs should be arriving from Adafruit tomorrow (Monday 2 March), so I should be able to try and replicate your results soon.  

For what it's worth, these sorts of issues (especially, the jitter you have noticed) are what motivated me to start working with hardware-based sensor-fusion solutions like the [USFS](https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-lsm6dsm-lis2md/)  from Pesky Products.  I have a [library](https://github.com/simondlevy/USFS) that works with the USFS on a variety of variety of MCUs and single-board computers.  I am currently designing a PCB to turn the USFS and the TinyPICO ESP32 board into a flight controller.  So unless you have a specific reason for staying with the Adafruit IMU, I'd highly recommend checking out the USFS.

danieladugyan, 2020-03-03:
In hindsight I've made some purchasing errors due to inexperience, and were I to redo my project I'd probably buy something like the USFS or the BNO055, and a STM32 instead of an ESP32. I've read something about adding a complementary filter to combat the drift I'm seeing when using the Madgwick filter, so I'm going to try and see if that helps.

simondlevy, 2020-03-04:
Okay, I figured it out: My software-quaternion code was expecting gyro in degrees/sec, but the NXP delivers rad/sec.  I fixed the bug, tested with the GCS, and committed the code.   It works as expected with Madgwick, although I'm still seeing the jitter you mention.

simondlevy, 2020-03-04:
Hey, as a reward for catching the degrees/radians bug in my code, I'd like
to send you one of the USFS units.  Do you have a shipping address I can
use?

I'm not sure if this message will show up on publicly on github, so to
reply to me privately: simon.d.levy@gmail.com.
======================

Simon D. Levy
Professor and Acting Department Head
Computer Science Department
Washington and Lee University
204 W. Washington St.
Lexington, VA 24450
https://simondlevy.academic.wlu.edu/

danieladugyan, 2020-03-05:
Thank you very much! Your code works as expected, and I figured out why I was getting that drift I talked about in my third post. I had changed `GYRO_MEAS_ERROR_DEG` to `0.f` , which yields zero jitter, but a lot of sensor drift. Experimenting with this value might be able to lead to a better middleground between jitter and drift. Currently I'm using `GYRO_MEAS_ERROR_DEG = 20.f`, which seems to be a bit better than `40.f`. @kriswiner commented on this value in the original [library](https://github.com/kriswiner/MPU9250/blob/master/MPU9250BasicAHRS.ino#L246).
I've also tried writing a new filter in [filters.hpp](https://github.com/simondlevy/Hackflight/blob/master/src/filters.hpp) based on Adafruit's filter, but it's still giving me very poor performance.
I was going to warn you about the degree/radians bug, but to be honest I forgot about it and I still don't quite understand everything in the code properly. 

- Why are you using `ACCEL_RANGE_4G` instead of `ACCEL_RANGE_2G` in [nxp.hpp](https://github.com/simondlevy/Hackflight/blob/master/src/imus/softquats/nxp.hpp)?
- Why doesn't a 9DoF filter give us better performance?
- Why doesn't [Adafruit's sensor calibration](https://github.com/adafruit/Adafruit_Sensor_Calibration) make any notable difference?

simondlevy, 2020-03-05:
Thanks; I've changed GYRO_MEAS_ERROR_DEG  from 40 to 20, and am indeed seeing less jitter.

To answer your questions:

* Why are you using ACCEL_RANGE_4G instead of ACCEL_RANGE_2G in nxp.hpp?
4G was in the original Adafruit sketches.  I changed my code to use 2G.

* Why doesn't a 9DoF filter give us better performance?
I fly indoors these days and so have avoided magnetometers and the issues surrounding their use.
Hence 6DOF instead of 9DOF.

* Why doesn't Adafruit's sensor calibration make any notable difference?
I can't say for sure, but if I had to guess, I'd say it's because of the 6DOF vs. 9DOF issue; i.e., it's mostly the magnetometer calibration that matters.

closed 2020-03-19 by danieladugyan