simondlevy/Hackflight #56: Basic Hackflight Compiling

MacOenghus, 2020-06-05:
Hello Dr. Levy,

I have been having trouble compiling Hackflight for the Ladybug Flight Controller using the Arduino IDE, and I was wondering if you could give me any advice.

I have:

-installed the latest version of the Arduino IDE, 1.8.12

-installed the gnu-arm compiler

I have been developing for the Adafruit Circuit Playground Express M0 (and M4) boards, so I think it works (fun board btw).

Then I:

-cloned the Hackflight repository into the ~/Arduino/libraries/Hackflight folder

-cloned the CrossPlatformDataBus repository into the
~/Arduino/libraries/CrossPlatformDataBus folder

-just in case, cloned the USFS repository into the ~/Arduino/libraries/USFS folder

-cloned the grumpyoldpizza repository into the
~/.arduino15/packages/arduino/hardware/grumpyoldpizza folder

No new boards show up in the Arduino IDE under Tools/Boards. I tried compiling BrushedMotorTest as a Adafruit M4 board and no luck. 

Any ideas?

Thanks,
Brian Dunlap

simondlevy, 2020-06-05:
Awesome username (MacOenghus), Brian!  I was seriously into Irish myth when I was in college, so kudos on the old-school spelling ;^)

Usually I'd try to reproduce an issue like this myself, but since the Adafruit Feather M4 board is currently out of stick, I think I can suggest something to try.  Since the board isn't showing up in Tools/Boards, my guess is that it has nothing to do with Hackflight.  You mention downloading the latest Arduino IDE, but did you follow Adafruit's [directions](https://learn.adafruit.com/adafruit-feather-m4-express-atsamd51/setup) for using the Board Manager to add the board to the IDE?

kriswiner, 2020-06-05:
Since this is the Ladybug flight controller (I assume) wouldn't it be more
appropriate to add the STM32L4 boards to the board manager (and the SAMD
board for the compiler)?

simondlevy, 2020-06-05:
I thought that Brian had been following the LadybugFC wiki guide for installing Hackflight but wanted to run it on the Adafruit M4 board instead.  Maybe I'm misreading.  Perhaps Brian could clear this up.

MacOenghus, 2020-06-05:
Thanks for the username comments, I think I've been using it since gmail was young (and experimental). And thanks for replying so quickly, although I may not be able to do much more this evening. I just received the Ladybug today, after ordering it on Tuesday!

simondlevy, 2020-06-05:
Great, so am I correct in assuming (with Kris) that you are attempting to compile for the LadybugFC, and not the Adafruit M4?

MacOenghus, 2020-06-05:
Yes, I just mentioned the Adafruit boards to confirm I had a working ARM compiler chain for non-Arduino boards.

simondlevy, 2020-06-05:
Sorry -- as usual, Kris figured it out before me! ;^)

Cloning my fork of grumpyoldpizza into the right place should make Kris's STM32L4 boards show up at the bottom of your Boards menu (so long as you re-open Arduino IDE after cloning).  On my Linux box I clone grumpyoldpizza into ~/Documents/Arduino/hardware, right next to ~/Documents/Arduino/libraries.  Since your Arduino folder is  ~/Arduino, for you it would be ~/Arduino/hardware.  Try closing the Arduino IDE, cloning grumpyoldpizza into ~/Arduino/hardware (you may need to make that folder if it doesn't already exist), reopening the IDE, and seeing whether the boards are there (at the bottom of the menu).

closed 2020-08-26 by simondlevy